salsanci/YRShell #8: make error linux

prof-braino, 2017-10-24:
I have build-essential installed.  Am I missing a step?

yrshell@yrshell-VirtualBox ~/Arduino/libraries/YRShell-20171022/build $ make
g++ -pthread  -c -o testall.o testall.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o testSupport.o testSupport.cpp -I . -I /usr/include -std=c++11 -O3
In file included from ../YRShell.h:7:0,
                 from testSupport.h:3,
                 from testSupport.cpp:2:
../utility/YRShellInterpreter.h: In member function ‘bool ShellRunner::init()’:
../utility/YRShellInterpreter.h:262:18: error: ‘virtual void YRShellInterpreter::init()’ is protected
     virtual void init( void);
                  ^
testSupport.cpp:9:18: error: within this context
     m_shell.init();
                  ^
Makefile:18: recipe for target 'testSupport.o' failed
make: *** [testSupport.o] Error 1
yrshell@yrshell-VirtualBox ~/Arduino/libraries/YRShell-20171022/build $

salsanci, 2017-10-24:
What commit are you working on? This should be fixed.

salsanci, 2017-10-24:
git rev-parse --verify HEAD

will give you the commit number you are currently using.

prof-braino, 2017-10-24:
newest; started fresh with git clone https://github.com/salsanci/YRShell.git
build essentials is installed.
braino@Aspire-VN7-793G ~/YRShell-20170918 $ gcc -o yrshell YRShell.cpp
gcc: error trying to exec 'cc1plus': execvp: No such file or directory

Am i missing a step?

prof-braino, 2017-10-24:
also tried

braino@Aspire-VN7-793G ~/YRShell/build $ gcc -o yrshell ./YRShell.cpp
gcc: error: ./YRShell.cpp: No such file or directory
gcc: fatal error: no input files
compilation terminated.

prof-braino, 2017-10-24:
ok, i messed this up. opening a new issue

closed 2017-10-24 by prof-braino