salsanci/YRShell #12: win 10 make error

prof-braino, 2017-10-27:
cygwin1@DESKTOP-LTS6E91 ~
$ cd c:

cygwin1@DESKTOP-LTS6E91 /cygdrive/c
$ ls
'$Recycle.Bin'  'Documents and Settings'   PerfLogs        'Program Files (x86)'   Recovery      'System Volume Information'   Windows
 cygwin64        pagefile.sys             'Program Files'   ProgramData            swapfile.sys   Users                        YRShell

cygwin1@DESKTOP-LTS6E91 /cygdrive/c
$ cd YRShell/

cygwin1@DESKTOP-LTS6E91 /cygdrive/c/YRShell
$ ls
build  examples  LICENSE  readme.md  rundocker.sh  utility  YRShell.cpp  YRShell.h

cygwin1@DESKTOP-LTS6E91 /cygdrive/c/YRShell
$ cd build/

cygwin1@DESKTOP-LTS6E91 /cygdrive/c/YRShell/build
$ make
g++ -pthread  -c -o testall.o testall.cpp -I . -I /usr/include -std=c++11 -O3
In file included from ../YRShell.h:4:0,
                 from testSupport.h:3,
                 from testall.cpp:3:
../utility/HardwareSpecific.h:23:23: fatal error: Arduino.h: No such file or directory
   #include "Arduino.h"
                       ^
compilation terminated.
make: *** [Makefile:18: testall.o] Error 1

cygwin1@DESKTOP-LTS6E91 /cygdrive/c/YRShell/build
$

salsanci, 2017-10-28:
Added a change which should support cygwin. If there are issues in utility/YRShellDefines.h
uncomment:

//#define COMPILE_DEBUG

and then

make clean

make

prof-braino, 2017-10-31:
Cygwin - make:
cygwin1@DESKTOP-LTS6E91 /cygdrive/c/Users/cygwin1/Documents/Arduino/libraries/YRShell-master-20171030/build
$ make
g++ -pthread  -c -o testall.o testall.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o testSupport.o testSupport.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o ../YRShell.o ../YRShell.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o ../utility/IntervalTimer.o ../utility/IntervalTimer.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o ../utility/Dictionary.o ../utility/Dictionary.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o ../utility/YRShellInterpreter.o ../utility/YRShellInterpreter.cpp -I . -I /usr/include -std=c++11 -O3
../utility/YRShellInterpreter.cpp: In member function ‘void YRShellInterpreter::beginParsing()’:
../utility/YRShellInterpreter.cpp:2230:48: error: ‘strtok_r’ was not declared in this scope
     m_token = strtok_r(m_Pad, "\t ", &m_saveptr);
                                                ^
../utility/YRShellInterpreter.cpp: In member function ‘virtual void YRShellInterpreter::slice()’:
../utility/YRShellInterpreter.cpp:2313:55: error: ‘strtok_r’ was not declared in this scope
             m_token = strtok_r(NULL, "\t ", &m_saveptr);
                                                       ^
make: *** [Makefile:32: ../utility/YRShellInterpreter.o] Error 1

cygwin1@DESKTOP-LTS6E91 /cygdrive/c/Users/cygwin1/Documents/Arduino/libraries/YRShell-master-20171030/build

prof-braino, 2017-10-31:
followed by:
cygwin1@DESKTOP-LTS6E91 /cygdrive/c/Users/cygwin1/Documents/Arduino/libraries/YRShell-master-20171030/build
$ make clean
rm -rf html
rm -f testall
rm -f yrshell
rm -f testresult
rm -f doxygen.log
rm -f *.o
rm -f ../*.o
rm -f ../utility/*.o

cygwin1@DESKTOP-LTS6E91 /cygdrive/c/Users/cygwin1/Documents/Arduino/libraries/YRShell-master-20171030/build
$ make
g++ -pthread  -c -o testall.o testall.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o testSupport.o testSupport.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o ../YRShell.o ../YRShell.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o ../utility/IntervalTimer.o ../utility/IntervalTimer.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o ../utility/Dictionary.o ../utility/Dictionary.cpp -I . -I /usr/include -std=c++11 -O3
g++ -pthread  -c -o ../utility/YRShellInterpreter.o ../utility/YRShellInterpreter.cpp -I . -I /usr/include -std=c++11 -O3
../utility/YRShellInterpreter.cpp: In member function ‘void YRShellInterpreter::beginParsing()’:
../utility/YRShellInterpreter.cpp:2230:48: error: ‘strtok_r’ was not declared in this scope
     m_token = strtok_r(m_Pad, "\t ", &m_saveptr);
                                                ^
../utility/YRShellInterpreter.cpp: In member function ‘virtual void YRShellInterpreter::slice()’:
../utility/YRShellInterpreter.cpp:2313:55: error: ‘strtok_r’ was not declared in this scope
             m_token = strtok_r(NULL, "\t ", &m_saveptr);
                                                       ^
make: *** [Makefile:32: ../utility/YRShellInterpreter.o] Error 1

cygwin1@DESKTOP-LTS6E91 /cygdrive/c/Users/cygwin1/Documents/Arduino/libraries/YRShell-master-20171030/build

prof-braino, 2017-10-31:
this does compile and run under linux, and windows 10 WLS (bash for windows),  but not on windows 10 using cygwin.

prof-braino, 2017-11-11:
use docker on windows

closed 2017-11-11 by prof-braino