salsanci/YRShell #28: byte, word, and lomng memory words?

prof-braino, 2017-11-14:
From caskaz:

We can find ! and @ in the word-list display using wl.
Are there plans for 16 bit  and 32 bit fetch and store memory operations?
Should these be called W@ and L@ as in PropForth, 
or perhaps D@ and Q@ (double and Quad) in the old examples; 
or should these be expressed in bytes such as 2@ and 4@?

prof-braino, 2017-11-25:
CAN't ACCESS general memory.  we can only access available registers. We can define how many we want. sal specifies in defailts #of registers - yrshell.h 16 reg my default, as is stack sizes.  

We can speifcy what text operations wee need, and do txt in the pad. 
cpystr, catstr, etc.  make a separate issue

prof-braino, 2018-01-17:
All memory access is limited to the YRShell defined registers by default.  This prevents errors from walking on the rest of the programs present.  All memory access is by fetch and those [forth words @ and ! ]  Fetch @ read a 32 bit register and puts it on the stack, and store ! takes a 32 bit value from the stack and put it into a 32 bit register. 
We can access byte (character) values using pad operations [ c@ and c!] c@ takes a value from the pad and pplaces it on the stack.  c! takes a value from the stack and places it at the current pad address.
We can display values from the stack to the terminal command line as byte values using .b 
We can display values from the stack to the terminal command line as 16 bit word values using .w

The default command to pop the top value from the stack and display it on the command line as a 32 bit value is the . command. (the period)

Put this in a wiki page

salsanci, 2018-01-17:
Does not use the pad, uses a separate text buffer. This prevents some bad
screwups.